bldrs-ai/Share #1: Extract reference coordinates from ifc

MarkusSteinbrecher, 2021-07-20:
In order to have a correct and precise reference for coordination between various ifc models, the architect defines a reference line/plane per building element (e.g. wall, slab, roof). This reference is stored in the ifc (as x, y, z coordinates and a vector) but this data is not easily extractable via common BIM/ifc-tools.

As a BIM coordinator I want to be able to use this information across various workflows and across various tools. The reference coordinates also enable me to get information regarding relationships of building elements (e.g. a wall connecting to another wall).

In a first step the reference coordinates could be extracted for all elements in a structured way (e.g. csv). In a next step, buildrs could offer an API that gives easy access to this information.

Example reference coordinates in ifc (file under buildrs/Share/src/Models/:

For this wall:

#2497= IFCWALL('3CLJkEIxT2IA4qBE524feg',#12,'tbd',$,$,**#2074**,#2456,'CC553B8E-4BB7-4248-A134-2CE142129A2A');

the reference coordinates are stored under:

**#2074**= IFCLOCALPLACEMENT(#144,**#2073**);

#2073= IFCAXIS2PLACEMENT3D(#2071,#2069,#2067);

here is the xyz coordinate (2071) and the vector (2067, 2069)

#2067= IFCDIRECTION((1.,0.,0.));
#2069= IFCDIRECTION((0.,0.,1.));
#2071= IFCCARTESIANPOINT((10.4294608578,24.9077803753,-0.139999999999));

So far the wall I get the point (10.42, 24.91, -0.14).

pablo-mayrgundter, 2021-07-22:
Use case:

Markus would like to perform a cost calculation to find the cost to join a bunch of walls.  He might select some walls, and then say, Which walls are connected to this initial set, and then depending on the wall types (e.g. wood to concrete), there will be different connectors with different costs.

Output should be CSV so Markus can join with other cost data in a way TBD.  

This could be a very specific wall-cost-calculator app, but there are so many instances of this.

More generally this would allow us to populate a graph DB that would allow us to ask questions in the search bar.

Use cases include: fire safety, energy, etc.

So overall, the original Feature Request contains a good idea for how to get this information, but the "answer" Markus is looking for would not have the coordinates included.. he just wants the relationships between the elements, and the grid system is the most clear and obvious way to get the relationships.

pablo-mayrgundter, 2021-08-06:
Ok, looks like these are showing up in props now.  This suffice for this request?

[image]

MarkusSteinbrecher, 2021-08-06:
looks awesome! only thing missing now is the vector info

pablo-mayrgundter, 2021-08-06:
They're there too.. poor formatting right now, but will clean up soon

[image]

OlegMoshkovich, 2022-08-17:
@MarkusSteinbrecher  @pablo-mayrgundter  this is great, but perhaps we should move it somewhere else.

MarkusSteinbrecher, 2022-08-17:
yes, to ifctool

OlegMoshkovich, 2022-08-17:
nice - i wonder if we can move the whole issue. will take a look.