bldrs-ai/Share #3: Extract building element information from ifc

MarkusSteinbrecher, 2021-07-21:
To enable information management for building elements we need to be able to extract the data from an ifc in a structured way. Depending on the source of the ifc, each element (e.g. walls, slabs, windows) contains a set of ifc property sets and attributes. 

As a BIM coordinator I want to be able to use this information across various workflows and across various tools. 

In a first step the information could be extracted by the user providing information on what elements and what property sets/attributes are needed. Ideally the tool would show an overview of the elements and property sets/attributes and the user could select the data for the extract. Additionally, this could also be part of an API for buildrs.

OlegMoshkovich, 2022-08-17:
@MarkusSteinbrecher @pablo-mayrgundter seems similar to the next issue. shall we move it into a document and close the issue?

OlegMoshkovich, 2022-08-18:
https://github.com/bldrs-ai/ifctool/issues/33 relevant issue

pablo-mayrgundter, 2022-08-18:
I agree it's related to the Reference Model issue.  It's good to move the other one to ifctool, but let's keep this for UX discussion about how to use the RM in Share.