bldrs-ai/Share #24: QR Codes for link sharing

pablo-mayrgundter, 2021-09-28:
The link sharing component should include an optional QR code

1. The QR code should embed the current URL, so that a user who scans the code will be brought back to where it was generated.
2. The QR can be opened separately and printed to PDF.
3. The QR code can be embedded in a given PDF, e.g. a 2D cross section plan in a PDF in the same repository.  This would enable a CI/CD workflow to publish plans to construction site production and bring the user there back to check if the on-site plans are the correct version for production.

Possible libs:

https://www.nayuki.io/page/qr-code-generator-library

OlegMoshkovich, 2022-08-17:
@pablo-mayrgundter this is good - but low priority at the moment - perhaps a wiki design document?

pablo-mayrgundter, 2022-08-17:
It's old but a good feature request.  I'd like to avoid having to track
state in multiple places.  It does need design when we get to it though

pablo-mayrgundter, 2022-08-17:
I think you had some screens for this too.  Could you link them publicly?  Maybe we can use this as a test bounty

OlegMoshkovich, 2022-08-17:
https://www.figma.com/file/xmkuP9IiGasxlrbfKRajVZ/BLDRS-Screens?node-id=1867%3A5285

this is the sketch of the flow. 
very low fidelity.

closed 2024-08-12 by OlegMoshkovich