bldrs-ai/Share #57: Version control (GitHub)🥉

pablo-mayrgundter, 2021-11-24:
https://docs.google.com/document/d/1JgLDfkghG6wqyYIplCQisSqyUK2XnRu43UEUEwAe8YI/edit#heading=h.ykol3e6pgo6s

fulcus, 2022-06-06:
Doc is currently private

pablo-mayrgundter, 2022-06-06:
Francesco, is that informative enough for now?

OlegMoshkovich, 2022-08-17:
@pablo-mayrgundter is this still relevant?

pablo-mayrgundter, 2022-08-18:
Thanks for the ping, it looks like the versioning in the URL does work but yeah, we still need a UI control to query for versions and make them selectable by the user.

oogali, 2022-09-08:
What is the definition of a `version`?

Are we referring to something specific, or are we referring to any valid input that Github accepts in its URL structure? (e.g. commit hashes, branches, _and_ tags)

OlegMoshkovich, 2022-09-08:
I think it is a branch.

OlegMoshkovich, 2022-09-08:
[image]

this is an example of a repo with two versions:
https://github.com/OlegMoshkovich/Logo/tree/main

Version 1:
https://github.com/OlegMoshkovich/Logo/blob/iteration_0/IFC_STUDY.ifc
https://bldrs.ai/share/v/gh/OlegMoshkovich/Logo/iteration_0/IFC_STUDY.ifc

Version 2:
https://github.com/OlegMoshkovich/Logo/blob/iteration_1/IFC_STUDY.ifc
https://bldrs.ai/share/v/gh/OlegMoshkovich/Logo/iteration_1/IFC_STUDY.ifc

oogali, 2022-09-09:
To be clear, you're referencing a repository with three versions.

One version in each branch:
- `main`
- `iteration_0`
- `iteration_1`

oogali, 2022-09-09:
Is there a Figma/mockup available for what the UI is expected to look like?

OlegMoshkovich, 2022-09-10:
Yes, you are right there are three versions.
High level mock up is here:
https://www.figma.com/file/LNUJP0228mJqlb4ev6AKbf/Design?node-id=4%3A4568

[image]

pablo-mayrgundter, 2023-02-20:
I talked with Bruno Postle @brunopostle, and he suggested ideas:

```
I wrote a proposal for a gui tool: https://community.osarch.org/discussion/comment/12598/#Comment_12598

Though I since elaborated this, but can't find where, some of the things not in the above proposal:

- isomorphic-git lets you connect to any repository, so there is no need to be github specific, github also has a hard 100MB limit on file sizes so you really don't want to be limited to github.
- Although working with single IFC files seems logical, a repository will allow multiple IFC files, so any git/IFC GUI will need to first point to a repository, then allow the user to browse IFC files in that repository. This makes more sense when you consider that multiple IFC files can be federated as an overlay.
- A true 3D visual diff between versions is hard, if not impossible, but a simple `git diff` will list all the modified and removed object GUIDs, which can be used to quickly colour the 3d view.
- three-way merges is advanced usage, a lot can be done simply with a git commit log and a tool to switch the 3d view between revisions.

..we even have a small amount of funding though osarch to work on this: https://community.osarch.org/discussion/comment/12981/#Comment_12981
```