bldrs-ai/Share #134: Portfolio for SP

pablo-mayrgundter, 2022-03-07:
We need to have a Image button preview of a screenshot of Tinyhouse, which when you click takes you to the GitHub path to the Momentum tinyhouse project.

This will also require some comments setup on their repo to work well.

But really for now we just need the button somewhere.

Maybe we tuck settings into an expando or accordion or move it back out into a separate ControlButton

OlegMoshkovich, 2023-01-12:
@pablo-mayrgundter don't fully understand this issue. 
Is it still relevant?

pablo-mayrgundter, 2023-01-13:
I think this is done.  We have images and links in comments, so functionality is there..  I may have wanted a link around images, but markdown supports that.  Good enough for now!