bldrs-ai/Share #149: Reload after uploading leads to an error

MarkusSteinbrecher, 2022-03-15:
When I upload a model from my local file system and hit refresh, the following error is shown:

[image]

I believe this is expected behavior, since the model was uploaded manually. Maybe we could just get rid of the error message?

pablo-mayrgundter, 2022-03-16:
Related to #114, the upload feature needs to be explained as a temporary view.. reloading not supported, but we need to block/warn about that.

OlegMoshkovich, 2022-08-17:
@pablo-mayrgundter is this still relevant?
[image]
in the disclaimer in bold at the bottom of the dialog, shall we add that reloading a page will lead to an error

pablo-mayrgundter, 2022-08-18:
I think what we have in the dialog is good, but we need to protect the user from inadvertently reloading as well.
- There's an HTML form/browser interaction thing where the browser knows you have unsaved state and will warn you on reload.  We should add that first.
- Once we have that, we can get of this error modal.

closed 2023-01-13 by pablo-mayrgundter