bldrs-ai/Share #169: Move site off GHP

pablo-mayrgundter, 2022-04-04:
This is required for handling Google Ads correctly as they don't like the 404 bounce.

I found this for howto setup the server:

https://gist.github.com/mjackson/b5748add2795ce7448a366ae8f8ae3bb

We also need a CD script to update the service on new pushes to main.

pablo-mayrgundter, 2022-04-05:
👀 https://www.mitrais.com/news-updates/aws-amplify-vs-google-firebase-which-is-better/
Leaning towards Goog, surprise surprise...

shafy, 2022-04-30:
Probably missing some context here, but I really like [Netfliy](https://www.netlify.com/). They even have a neat feature that builds a preview and links to it for every PR automatically. Or do you want to move away from static hosting?

oogali, 2022-06-14:
This issue is superseded by #224.

closed 2022-06-14 by oogali