bldrs-ai/Share #189: Load model menu has wrong tooltip and icon

shafy, 2022-04-29:
The load model tooltip reads "Shortcut keys", and the used icon is usually associated with sharing, not opening a file.
[image]

I suggest using the folder icon that's already used in the modal:

[image]

pablo-mayrgundter, 2022-05-03:
Looks good to me.  @OlegMoshkovich agree?

OlegMoshkovich, 2022-05-04:
yes sounds good!
change is going out now.

shafy, 2022-05-07:
Looks like this is deployed. Closing.

closed 2022-05-07 by shafy