bldrs-ai/Share #198: Setup: Failure to build & serve on Win11

Adrian62D, 2022-05-12:
**Platform:**
- Windows 11, 22000.675
- node v16.15.0
- npm v8.5.5
- Firefox v100.0

**Steps to reproduce:**
1. Follow the [setup instructions](https://github.com/bldrs-ai/Share) i.e.:
2. Open a command prompt
3. `yarn install`
4. `yarn serve`

**Expected result:**
- Browser opens http://localhost:8080/
- bldrs is shown

**Actual result:**
```
>yarn serve
yarn run v1.22.18
warning package.json: No license field
$ yarn build && node config/serve.js
warning package.json: No license field
$ yarn copy-web-asm && yarn clean && yarn new-version && node config/build.js
warning package.json: No license field
$ cp node_modules/web-ifc/*.wasm public/static/js
'cp' is not recognized as an internal or external command,
operable program or batch file.
error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.
error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.
error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.
```

**Details:**
- package.json appears to contain platform-specific shell commands
- When patching those commands (workaround) the `yarn serve` command succeeds. However, http://localhost:8080/ shows a directory listing with 4 files (docs/) instead of loading bldrs:

[image]

Thanks for your support.

BR,
Adrian

Adrian62D, 2022-05-13:
Hi all

After applying further changes to package.json I got it to launch properly in the browser. Great.

For sake of completeness, the current state of the windows-workaround below:

```
  "scripts": {
    "clean": "rd /s /q docs & xcopy public\\ docs\\ /E /H /Y",
    "new-version": "node --experimental-json-modules src/utils/version.mjs > package.json.new && move package.json.new package.json",
    "copy-web-asm": "xcopy node_modules\\web-ifc\\*.wasm public\\static\\js /E /H /Y",
    "build": "yarn copy-web-asm && yarn clean && yarn new-version && node config/build.js",
    "serve": "yarn build && node config/serve.js",
    "lint": "yarn eslint `find src -name '*.jsx'` `find src -name '*.[m]js'`",
    "test": "jest",
    "precommit": "yarn lint && yarn test",
    "prepare": "husky install"
  },
```

Maybe [shx](https://github.com/shelljs/shx) could be good solution to achieve cross-platform support?

BR, Adrian

OlegMoshkovich, 2022-05-16:
Adrian, 
Thank you so much for the work around!
Since the issue is resolved I will close it, and move the proposed solution to the wiki.
Oleg

pablo-mayrgundter, 2022-06-01:
@Adrian62D here's a mostly fix for ya!

Adrian62D, 2022-06-02:
Thanks a lot, I'll try it out.

pablo-mayrgundter, 2022-07-25:
I think this was fixed.  @Adrian62D please reopen if not.

closed 2022-07-25 by pablo-mayrgundter