bldrs-ai/Share #199: Support integration of Bldrs at the DOM level

Adrian62D, 2022-05-12:
As a developer I want to integrate Bldrs at the DOM level so that I can optimize the experience for the end-user.

Background:
- Currently in our prototype Bldrs is integrated via iframe which is not an optimal solution.
- It would be great if Bldrs can be instantiated programmatically and be configured to render to a custom DOM element target.
- This could potentially be implemented as a js/ts library or as a set of wrapper library for angular/react/vue.

BR, 
Adrian

pablo-mayrgundter, 2022-06-01:
We've done an in-person planning session with Adrian on this.  Thanks Adrian!

Here's some notes:

- Bldrs itself uses the URL for as much state as reasonably possible, to achieve high-utility permalinks.  Generally, cross-component interaction that results in UX state should be captured in the URL.
- Bldrs intends to enable an app developer to control large parts of the UX, so we need to be careful about how embedding happens.  Arbitrary embeddings and page state control seem difficult to ensure.
- An iframe may be preferable in general for the process and security isolation.  Here's some background on how Chrome has been looking at it:
https://www.chromium.org/developers/design-documents/site-isolation/

All that said, it's definitely *not* off the table for us.  I think it's more priority for now: iframe > apps > embedding.  We're also open-minded on this, especially for different opinions on the points above.

OlegMoshkovich, 2022-08-17:
@pablo-mayrgundter what would you like to do with this issue?

pablo-mayrgundter, 2022-08-18:
@MarkusSteinbrecher Could you bring this up-to-date given our discussions with Vyzn?

MarkusSteinbrecher, 2022-08-25:
This topic will be discussed and solved separately. Closing this issue.

closed 2022-08-25 by MarkusSteinbrecher