bldrs-ai/Share #200: Support integrating Bldrs optimized for performance and costs

Adrian62D, 2022-05-12:
As a developer I want to ensure performance (low latency, low bandwidth consumption) and cost optimization when integrating Bldrs in my SPA.

Background:
- The current data exchange from/to Bldrs is designed with a persistence layer for github. This is a nice, interoperable approach.
- In addition, from an SPA perspective it would be nice to also have an approach that is geared towards performance and cost optimization e.g.:
  - load, modify and dispose one or more building models in the context of the SPA 
  - request the Bldrs component that runs within the SPA context to render any of these building models 
  - This would allow the SPA to control precisely when and how data is transferred over the wire (bandwidth optimization, serialization techniques)
  - This would allow the SPA to support immediate switching between different models or variations
  - This would allow the SPA to modify the models i.e. display custom content and/or run computationally expensive algorithms on the client side instead of using resources in the cloud (scalability)
  - This would allow the SPA to prevent intermediary persistence calls (cloud costs, latency)

Related to #199.

BR,
Adrian

pablo-mayrgundter, 2022-06-02:
Thanks for more detail on your design goals here.

If I bracket out the solution of handling within the SPA, then we already handle these:

- load, modify and dispose one or more building models[ in the context of the SPA]
- request the Bldrs component [that runs within the SPA context] to render any of these building models
- immediate switching between different models or variations

which leaves this new functionality we could perhaps address:
- control precisely when and how data is transferred over the wire (bandwidth optimization, serialization techniques)
- modify the models i.e. display custom content and/or run computationally expensive algorithms on the client side instead of using resources in the cloud (latency, costs, scalability, storage)
- prevent intermediary persistence calls (costs, latency)

Do you have any more thoughts on types of bandwidth or serialization control you'd be looking for?

For model modification, we're expecting to support a RESTful API to elements identified by their paths on the URL.  

Expensive compute will probably be multi-process too, so I don't see a particular win for including in the single page.  I'd appreciate more of your thoughts here.

Thanks!

Adrian62D, 2022-06-02:
> If I bracket out the solution of handling within the SPA, then we already handle these:

To clarify, do you mean one of the following approaches?

- The SPA instantiates multiple iframes (of Bldrs) and then shows/hides them dynamically
- The SPA instantiates a single iframe (of Bldrs) and Bldrs would support managing multiple building models in a single instance

> Do you have any more thoughts on types of bandwidth or serialization control you'd be looking for?

My assumption is that the performance perceived by the end-user is a success criteria for the SPA. And so the more control the developer of the SPA has over the 3D viewer component the more options are available to optimize the SPA. In terms of bandwith and serialization the developer could control exactly what versions of the model to keep in the clients memory, modify/patch the models in-memory as the user utilizes the SPA and perform a network-roundtrip to (any) API only when needed.

Reflecting on this topic a bit more: The basic conflict of interest here is that from Bldrs perspective you strive for abstraction and isolation while as an SPA developer you strive for control. For example, the interfaces provided by ifc.js provide fine granular control and are more suitable in a SPA scenario than the abstraction layer provided by Bldrs. This is not a criticism, it's an observation.

> Expensive compute will probably be multi-process too, so I don't see a particular win for including in the single page. I'd appreciate more of your thoughts here.

We are currently implementing the algorithms in the backend. Among a set of other reasons this allows us to ensure that data is not corrupted so that users can not create fake energy certificates. With every new app we build there will be a new algorithm. Some are simple, some are more complex. My assumption is that if have a significant number of users and a set of expensive algorithms then we would want to have the option to offload that work to the client devices instead of draining our cloud budget. The problem is not that we cannot handle a handful instances of the algorithm, the problem is the situation when you have a significant number of users. It is maybe comparable to your decision to have the user provide his/her own storage (Github, Google Drive) instead of investing your cloud budget to provide this.

BR, Adrian