bldrs-ai/Share #202: Bldrs is not served from Github Pages when forked

Adrian62D, 2022-05-16:
**Steps to reproduce:**
1. Fork Bldrs to a repo that is **not** named "Share"
2. On the fork configure[ Github Pages](https://docs.github.com/en/pages/getting-started-with-github-pages/configuring-a-publishing-source-for-your-github-pages-site#choosing-a-publishing-source) to serve the /docs directory
3. Access the URL from the Github Pages with the browser

**Expected:**
- The page is shown in the same way as [https://bldrs-ai.github.io/Share/](https://bldrs-ai.github.io/Share/)

**Actual:**
- The page does not load

**Cause:**
There is a hard-coded repo name ("Share") in docs/index.html:

```
<script>
  const loader = document.createElement('script');
  const pathPrefix = window.location.pathname.startsWith('/Share') ? '/Share' : '';
  loader.setAttribute('src', pathPrefix + '/index.js');
  document.body.appendChild(loader)
</script>
```

BR, 
Adrian

pablo-mayrgundter, 2022-06-04:
I've tried a couple workarounds for this, but I don't think there will be anything automatic without much deeper understanding of GitHub API.

We're planning to move staging to a separate hosting service.

In the meantime, a bit of code simplification and a configuration guide for the installation path will suffice.

pablo-mayrgundter, 2022-07-25:
I think this was fixed.  @Adrian62D please reopen if not.

closed 2022-07-25 by pablo-mayrgundter