bldrs-ai/Share #207: Tests fail on M1 Mac with Node 16+

pablo-mayrgundter, 2022-05-24:
I've been testing moving to node 16 and 18.

Node 14 works for me, but doesn't support new Mac M1 binaries, so everything was being run in emulation mode.. that explains the speedup for me when I switched to node 16.

However, if i rebuild node_modules from scratch with 16 or 18, they now expect custom M1 binaries.  Taking canvas and gl out of the package solved the install, but unfortunately it looks like there's a dependency back to canvas  from the testing framework.. all tests fail with variations on:

```
$ jest
 FAIL  src/Components/CameraControl.test.jsx
  ● Test suite failed to run

    dlopen(/Users/pablo/node_modules/canvas/build/Release/canvas.node, 0x0001): tried: '/Users/pablo/node_modules/canvas/build/Release/canvas.node' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e'))

      at Object.<anonymous> (../../node_modules/canvas/lib/bindings.js:3:18)
 ``

I tried adding canvas explicitly back to package.json, but am not getting the pre-built binaries :/. I tried building the binaries, but the build fails with some sort of misconfig of my environment looking for C++ libs that i do have installed but node-gyp can't find
so, currently I have tired of shaving this yak and will now return to 14.

shafy, 2022-05-25:
I've figured it out. The issue is related to `canvas` and you need to install `cairo` and `pango` (`brew install cairo pango`) before running `yarn install`. It then build `canvas` correctly without relying to download the missing libraries. See [the open issue here](https://github.com/Automattic/node-canvas/issues/2038) (probably good idea to follow it and see when this is fixed).

I've added a bit to the readme and submitted a PR: #208

oogali, 2022-06-14:
I believe there were two issues at hand: 1) lack of the proper runtime dependencies, and 2) Pablo's laptop had a special installation of Brew in his home directory rather than at the upstream default (of `/opt/homebrew`).

The former prevented anything from running, but the latter made it impossible for Pablo to link against the libraries after they were installed on his system.

The `canvas` project contains a build script that attempts to find the correct dependencies, but it is hard-coded to only use the Brew default installation directory of `/opt/homebrew`. If anyone has a non-standard installation location, `canvas` will not use it... leading to tears and sadness.

closed 2022-06-14 by oogali