bldrs-ai/Share #223: Enable React Router compatibility

oogali, 2022-06-14:
When using the `BrowserRouter` component of the React Router library, a general issue frequently encountered is if a user takes an application deep link and attempts to pull that up in their browser, they will receive a 404 Not Found error from the serving web server.

The 404s are caused by the web server not knowing that these additional virtual paths are all handled by the same React SPA.

This gap can be solved by configuring the web server appropriately to always return the HTML of the SPA in response to any request for a [valid] route.

Once this is solved, the expected result is that deep linking into BLDRS should no longer raise a 404 Not Found error to the browser.

OlegMoshkovich, 2023-02-03:
@oogali is this in progress?

pablo-mayrgundter, 2024-04-23:
This was implemented a long time ago and changed desc to note we still use 404 bounce for cypress and dev.

closed 2024-04-23 by pablo-mayrgundter