bldrs-ai/Share #239: Poor performance with large models

MarkusSteinbrecher, 2022-06-23:
The performance with large models is currently not what the user expects. The frame rate is low when rotating the model and clicking an element takes multiple seconds to update (while the screen is frozen). Information on properties takes multiple minutes to load.

Tested with this model: https://drive.google.com/file/d/1-5Ufg8PmDTr6GFGzOLQcgK6SM2jn_Vm-/view?usp=sharing

pablo-mayrgundter, 2022-06-25:
Can we publish this model?  With the LFS fix we should be able to serve it in prod

MarkusSteinbrecher, 2022-08-17:
You mean the issue description needs more detail?

OlegMoshkovich, 2022-08-17:
I think optimization of Large models will touch many parts of the system, and at some point when we are ready for this work, mode detailed issue list will emerge out of design/discussion.

pablo-mayrgundter, 2022-08-18:
I think there's 2 parts to the slowness:
  - framerate
  - UI component responsiveness

I don't think we'll do much about framerate in the near-term, but I also think Markus is more sensitive about this than your average user! ;)

I read this as UI component responsiveness, and it does take time to select items in the UI.  We should work on this.  I think this scope is fine.  I'll reword the title and @MarkusSteinbrecher, please try a large model again and see if framerate is still low.  If so please open a separate bug for that.

MarkusSteinbrecher, 2022-08-18:
well if you give me WASD I need at least 120 fps ;)

MarkusSteinbrecher, 2022-10-11:
I think it should be on publig github already:

https://github.com/Swiss-Property-AG/Seestrasse-Public

OlegMoshkovich, 2022-12-23:
Another Example of the slowness is trying to select an element on this model
https://bldrs.ai/share/v/gh/Swiss-Property-AG/Turmfalke-Public/main/TURMFALKE.ifc/88/111/138/1898885/1916909#p:y=2.3394265647800356::c:19.31,20,-14.82,2.55,8.47,5.74 
noticeable latency in response.