bldrs-ai/Share #248: Load multiple ifc into a scene

MarkusSteinbrecher, 2022-07-06:
As a user I want to be able to upload multiple ifc models into a scene and be able to toggle each (e.g. architecture model and HVAC model).

OlegMoshkovich, 2022-07-19:
@pablo-mayrgundter - does the work you did with the cars, address this issue?

pablo-mayrgundter, 2022-07-26:
No.  I think we want to use the IFCjs loader to handle multiple models.  Let's leave this open to track that.

MarkusSteinbrecher, 2022-08-24:
Apparently this is already possible with ifc.js: https://stackoverflow.com/questions/68222208/how-to-load-multiple-files-in-ifc-js

MarkusSteinbrecher, 2022-09-07:
Updated the description. I propose to start with the easiest version in the UI and just add a "add model" to the "Open IFC" screen. What do you think?

pablo-mayrgundter, 2022-09-08:
I think we have two related concepts here: Levels of a building and Layers of a building.  "Show me MEP & Architecture for level 4 of the building" includes both of these.

I propose 2 tool icons (on right side), one for each:
  - Section tool has the usual stack of squares, and allows toggling floors on and off.  
  - Layer tool is next to it, different icon, and allows toggling e.g. MEP & architecture on and off.
    - Layer tool also includes a "+" symbol, which triggers a file upload of an IFC that creates a new layer, that can also be toggled.

Separately, loading files in the top-level search dialog area is always "start a new project by removing all current content", or perhaps there's a checkbox to leave current content.  However, that raises the problem of whether the new content is in its own layer, and if so why not use the layer tool approach.

Talked with Oleg on this.. think we need some mocks.

MarkusSteinbrecher, 2022-09-08:
Not sure I agree to the "layer tool" ... this is about loading another ifc model into a scene. The "layer tool" in my view would be one ifc model but I mark MEP as a layer inside that. Maybe this is just naming but if I want to upload another ifc into a scene I am not searching for "layer tool".
Regarding the mock ups: I am sure we can come up with a nice ui over time ... for now adding a "load additional model" on the open screen would work I think.

pablo-mayrgundter, 2022-09-13:
Ok, as discussed, this is really just loading top-level models, i.e. with separate trees, no particular merging of layers (how do we know what's a layer?).  Good good

Adrian62D, 2022-12-02:
Fits to product strategy: yes
Estimation: needs R&D work for Layering, potentially Q1

Adrian62D, 2023-04-21:
@pablo-mayrgundter do you see an issue if we start with this one already even though the core refactoring on your end is in progress?

pablo-mayrgundter, 2023-04-24:
Really depends how complex this will be.  If it's pretty simple, then it's ok as we're not committing too much work before the new datamodel is active.  So I'd suggest a day or two looking at it to see if there's a quick approach.

Adrian62D, 2023-05-05:
We analyzed the requirements and updated the issue description accordingly. We think it's doable with an OK-ish amount of effort and will tackle this one.

pablo-mayrgundter, 2023-05-09:
Good to hear.

IFCjs has support for this, and we'll be following that API for the new datamodel.