bldrs-ai/Share #307: Notes should match current model

pablo-mayrgundter, 2022-08-03:
In notes, we currently show all issues in the repo, independent of what model you're viewing.

I propose we show only issues somehow related or tagged to the currently displayed model.

This also suggests that if we could display a whole repo's set of models (i.e. for Portfolio, or directories of models per Greg's feedback) then we'd display all issues for the directory and its subdirectories.

pablo-mayrgundter, 2022-08-03:
For example, I have 3 issues in this repo (a fork of ifctool).. if I do a search for the whole path of the model as linked in the first issue, then it matches the first issue and third, and not the second issue, even though the second issue does contain the filename.

The first and third issue have the full path in them.  The full path is visible to the user in the first issue, but hidden as a link in the third.

So this would be somewhat flexible in bringing up any issues matching the current file as long as the full path to it in bldrs is linked.

OlegMoshkovich, 2022-08-10:
@pablo-mayrgundter what is the status on this?

pablo-mayrgundter, 2022-08-30:
Shelved it a bit to work on testing.. will dust off soon