bldrs-ai/Share #309: User defined size of images in notes

MarkusSteinbrecher, 2022-08-03:
Currently the image size in notes seems to be fixed. If I want to upload a smaller image, it will be blown up. As a user I want to either set the size of the image before I upload it to Github, or be able to resize it in the notes.

OlegMoshkovich, 2022-08-04:
it is a good one but a bit challenging.
how would you set the size in the note itself when you create a note?

MarkusSteinbrecher, 2022-08-04:
Can we use the original image size?

OlegMoshkovich, 2022-08-10:
At the moment the image is styled based on the width of the card, the height of the image is calculated automatically
I propose to define a constant dimension for the height 
- let's say 500px.
In the note footer, we can include a flag
‘Prioritize Height‘= true.

If the height is prioritized the width becomes variable.

[image]

OlegMoshkovich, 2022-08-17:
will get to this one this sprint