bldrs-ai/Share #310: Toggle Notes Feature

MarkusSteinbrecher, 2022-08-03:
As a user I want to toggle the content of notes (show less/show more), similar to what we have with properties.

pablo-mayrgundter, 2022-08-03:
How is this different from #306?

MarkusSteinbrecher, 2022-08-03:
#306 is always expand and has no functionality of toggle on/off and since we have #306 implemented already it is done anyway I guess

MarkusSteinbrecher, 2022-08-04:
the "show more / show less" was not ideal, because it cuts the note in a weird way.

"show more / show less" should show all of the note (=show more) or only the title (=show less).

In addition to that, there is a toggle button on top of notes to do that - just not only for a single note but for all of them.

pablo-mayrgundter, 2022-08-04:
Ok, I get it.

Toggling issues makes sense when multiple issues are listed, and they will have a title.

For the comments on the issue, there's no title per se, but clicking the metadata header to toggle it would work.

OlegMoshkovich, 2022-08-10:
This to me is a feature request. 
I propose putting it into a backlog on the notes project.

OlegMoshkovich, 2022-08-17:
I am not sure how relevant this is anymore. since we are adding multiple images to the issues. will close the issues, and we can reopen - later if we still think that it is relevant.

OlegMoshkovich, 2022-08-17:
Ok reopening it 
@MarkusSteinbrecher what do you think - is more/less still relevant?
I think the issues as they are now are quite good, in my opinion, is it better to see all of the information and work on the way to filter the issues.

MarkusSteinbrecher, 2022-08-25:
I think it is still a nice feature, although not the highest priority. If you have a lot of notes it may be easier to "show less" and scroll through them to find what you are looking for.