bldrs-ai/Share #311: User defined order of notes

MarkusSteinbrecher, 2022-08-03:
At the moment, the order in which notes are shown is defined by the order in which they are created in Github. That means that updates on notes are no fun ... As a user I want to be able to define the order of the notes - possibly through some metadata in the note itself (e.g. #nr-1)

pablo-mayrgundter, 2022-08-03:
How about the sorting available from github? 
[image]

MarkusSteinbrecher, 2022-08-03:
What if I create 10 issues and then decide at the end that I need another one in the middle? I don't think the standard sorting is enough for this.

pablo-mayrgundter, 2022-08-04:
Ah, that sounds like the portfolio use-case where you're really trying to present information in a certain order.  Maybe we can integrate these with a single solution

OlegMoshkovich, 2022-08-04:
@pablo - do you oppose implementing a meta string in the note itself?
'possibly through some metadata in the note itself (e.g. #nr-1)'

pablo-mayrgundter, 2022-08-05:
Yeah, that's reasonable.  Just want to make sure we keep the use-cases differentiated.

Like, let's say we do this so that we can achieve a presentation function.  That's cool for presentations, but then someone might come along and look at a set of notes and expect them to be sorted by post-time, i.e. as a comment thread.  At this point, the explicit ordering becomes complicated to think about.  How does the user control it?  How do they know if it's active?

So I propose prioritizing the development the modal switch.  A user should be able to use comments to build a presentation, but there's a clear modal shift to switch into presentation and out of it.  And for this feature, when not in presentation mode, the comment ordering tags would be ignored.

OlegMoshkovich, 2022-08-10:
Feature Request.
Putting it into Notes backlog

ronplaymeta, 2023-03-30:
It seems like good to use GitHub itself sorting feature too with presentation mode by using modal switch.