bldrs-ai/Share #313: Notes: issue panel state should be in permalink

pablo-mayrgundter, 2022-08-03:
Toggling issues open and closed clears the hash params, which seems like a regression.  @OlegMoshkovich was this working at some point?

Either way, toggling issue panel open and closed should keep state in the hash params.

pablo-mayrgundter, 2022-08-05:
@OlegMoshkovich it looks like camera state is being cleared from permalink when notes are opened and there's not a selected note id.  Do you know why?  Looks like the change was in your #235.

https://github.com/bldrs-ai/Share/blame/main/src/Components/IssuesControl.jsx#L32

OlegMoshkovich, 2022-08-06:
When the user moves from a note with a camera defined to a note with no camera defined if we don't clear the URL and the user decided to share the note - a camera position from the previous note is shared.

pablo-mayrgundter, 2022-08-10:
So I think that's just preferable.

The invariant should be that the URL expresses what you see.  If you move to a new note with a camera defined, but you don't click on it on the camera button, then you don't see the new view.  The invariant is preserved: the view you see is still expressed by the current URL.

Removing the clearing behavior on component load also allows the the issue panel to be toggled without affecting URL.

If we want the action of selecting a new issue or note to change the view, I think maybe there's some different feedback for that.  Maybe we make the camera icon non-interactive and have it represent just the presence of a camera, and the camera is activated when the issue or note is clicked.  If we do this, I think we still delay clearing the camera until an issue or note becomes active.

closed 2022-08-17 by OlegMoshkovich