bldrs-ai/Share #314: Notes not updating if link to ifc on Github is pasted into search

MarkusSteinbrecher, 2022-08-04:
Go to Bldrs homepage, open notes and then paste a link to an ifc into the search ... new ifc model is loaded but the notes do not update:

[image]

OlegMoshkovich, 2022-08-04:
yes. I think it is happening because we are not triggering issue pull on the model load.

OlegMoshkovich, 2022-08-10:
Problem: it is happening because the user loads the new project while on the "SINGLE NOTE" view in the current project, 
When the new project is loaded the specific note cannot be located.

Proposed solution: Change the "SINGLE NOTE" view to the "LIST of NOTES" view automatically when the new project is loaded.

closed 2022-08-17 by OlegMoshkovich