bldrs-ai/Share #315: Notes navigation incorrect after reloading

MarkusSteinbrecher, 2022-08-04:
Opening the notes and skipping through them from 1/5 to 5/5 works fine. If you are on note 5/5 and reload the page, the display changes to 1/5 although it is still showing note 5/5

[image]

MarkusSteinbrecher, 2022-08-04:
tested here: https://deploy-preview-308--bldrs-share.netlify.app/share/v/gh/Swiss-Property-AG/Astra-Public/main/ASTRA.ifc#i:1328337651::c:67.99,25.75,15.54,17.3,5.44,8.64

OlegMoshkovich, 2022-08-04:
@pablo-mayrgundter @MarkusSteinbrecher this has to do with the way we are keeping track of the state - reloading the page is causing the state to clear.

The number count is really nice and adds clarity.
However, dues to the complexity I am going to remove it in this iteration.
[image]

OlegMoshkovich, 2022-08-04:
@pablo-mayrgundter unless you know a straightforward way to fix it.

pablo-mayrgundter, 2022-08-05:
I think storing in the permalink is the right way forward, and made a similar comment in #319.

OlegMoshkovich, 2022-08-10:
Feature request. 
At the moment the issue is solved by not showing the note number in the controls.

MarkusSteinbrecher, 2022-08-10:
OK, now it is a feature request

OlegMoshkovich, 2022-08-10:
[image]
in the current iteration - the issue is solved by removing the not index from the controls.

closed 2022-08-17 by OlegMoshkovich