bldrs-ai/Share #318: Notes - Element selection only works if tree view is open

MarkusSteinbrecher, 2022-08-05:
I created notes for this project: https://bldrs.ai/share/v/gh/Swiss-Property-AG/Seestrasse-Public/main/SEESTRASSE.ifc/86/109/136/28708/41077

On note Nr3, the slab on the ground floor is selected and included in the permalink.

- if the tree view is selected, the element is selected already on the first note
- if the tree view is hidden, the element is not selected at all

Does element selection require the tree view to be open?

MarkusSteinbrecher, 2022-08-05:
also: if I select any element (e.g. the roof) while I have the notes open; then clear selection and skip to the next note, the roof is selected again.

OlegMoshkovich, 2022-08-05:
Hello.

OlegMoshkovich, 2022-08-10:
This is a general bug that has been there since the beginning, not related to Notes.

pablo-mayrgundter, 2022-08-15:
I'm wondering if this is still happening after #328?  I can now click around on different comments' cameras and not reset what element is selected.  This is probably a problem itself, but not seeing any interaction with tree being open or not.

MarkusSteinbrecher, 2022-08-15:
it seems to be working fine now, I can select elements without the tree

pablo-mayrgundter, 2022-08-16:
Cool.  Ok, opening different issue for that.. it's simpler too.

pablo-mayrgundter, 2022-08-30:
Hey @MarkusSteinbrecher could you review and close if fixed now.

closed 2022-08-31 by OlegMoshkovich

MarkusSteinbrecher, 2022-08-31:
it is still the same behavior as I commented above: "it seems to be working fine now, I can select elements without the tree. What is still there is this: Open notes view and select an element; deselect the element with the "clear selection" item; go to the next note and the element is selected again."

Selection works if tree view is closed. But if you select and deselect an element and skip through notes, the element is selected again. If this issue was just about the tree view, I guess it can be closed and we can create another issue.