bldrs-ai/Share #319: Notes should hide control items if placed under a footer

pablo-mayrgundter, 2022-08-05:
We scan URLs in notes to enable a camera button.  We may do more things like this as well.

It would be nice to hide this from the user at first glance, but make it easily discoverable.

I propose we use a separator like in GMail, e.g. '--', that demarks the beginning a footer section.  The separator will need to be compatible with Markdown, i.e. not get special rendering or have a rendering compatible with being a footer.

I tried testing just now.. not sure if it's as I remember, where the user can type it.. but the '...' here has the same role:

[image]

Whereas a current note with Markdown camera support looks like this:
[image]

The footer separator should hide the View link, until clicked.

OlegMoshkovich, 2022-08-10:
Feature Request.

OlegMoshkovich, 2022-08-17:
this is really good. 
@pablo-mayrgundter assigned it to you, if it is ok.