bldrs-ai/Share #337: zustand test warnings

pablo-mayrgundter, 2022-08-18:
Many zustand tests currently generate a warning as below.  Not sure how to solve, but may require refactoring zustand tests to just mock zustand instead of using it.

```
Warning: An update to TestComponent inside a test was not wrapped in act(...).

      When testing, code that causes React state updates should be wrapped into act(...):

      act(() => {
        /* fire events that update state */
      });
      /* assert on the output */

      This ensures that you're testing the behavior the user would see in the browser. Learn more at https://reactjs.org/link/wrap-tests-with-act
          at TestComponent (/Users/pablo/c/pm/Share/node_modules/@testing-library/react/dist/pure.js:281:5)
```

pablo-mayrgundter, 2022-08-18:
@OlegMoshkovich could you take this?  I tried debugging this but figure you'll have a better time of it.

OlegMoshkovich, 2022-08-18:
yes i can this one

pablo-mayrgundter, 2022-08-19:
I think it's possible the use of renderHook here is not as intended.. check this:
https://stackoverflow.com/questions/64287448/react-hook-testing-with-renderhook

pablo-mayrgundter, 2022-08-20:
I'm perplexed.. I banged my head on this for awhile.

I made a new repo to just test zustand with react hooks.  Everything works fine.

https://github.com/pablo-mayrgundter/react-zustand-jest

pablo-mayrgundter, 2022-08-20:
I've whittled it down to just this causing the warning in SideDrawer.test.jsx, first test, and surprisingly, it's the expect line that triggers the warning:

```
test('side drawer notes', async () => {
  render(<ShareMock><SideDrawerWrapper/></ShareMock>)
  expect(await screen.findByText('Notes')).toBeInTheDocument()
})
```

I read up on this:

https://kentcdodds.com/blog/fix-the-not-wrapped-in-act-warning

Current theory is that we have async code resolving after the expect, and testing-lib is telling us with the warning.  I suspect it's loading the issues async somehow.. so it may be like needing a findBy after render that looks for something in the issues.

Better yet would be to mock out the async results with something synchronous.  Would speed up tests too.

pablo-mayrgundter, 2022-08-26:
https://github.com/pmndrs/zustand/issues/363#issuecomment-845536384

"if you have an effect that updates the store, then the problem will happen again".  Sounds like us!

closed 2022-09-08 by OlegMoshkovich