bldrs-ai/Share #357: BCF support in Notes

MarkusSteinbrecher, 2022-08-26:
Adding BCF support in Notes is a feature most users will expect and it would also be useful for Swiss Property model quality checking ([user requirements](https://docs.google.com/document/d/1mtc1ZakbEwQNmYxHvIIqZ8o20Cs4X1CBL6x2OxxBe7I/edit#heading=h.v1dfzomgpzyj)).

OlegMoshkovich, 2022-09-06:
Need more requirements

MarkusSteinbrecher, 2022-09-07:
Draft requirements are documented [here](https://docs.google.com/document/d/1v_orXmYwlgFSGTxx3YO_SjVYPLwsb9koBhrzNJ2OXhU/edit#)

Adrian62D, 2022-12-02:
Fits to product strategy: yes
Effort: needs R&D work

Adrian62D, 2022-12-08:
@MarkusSteinbrecher Could you make the requirements document public? We would like to discuss it with the team. Thanks.

Ibrahim5aad, 2023-01-09:
Found this package, but only provide reading/parsing of BCF files.
https://github.com/Parametricos/BCF-js

Adrian62D, 2023-01-17:
Updated the issue description.