bldrs-ai/Share #380: Selection of one or more elements

Adrian62D, 2022-09-15:
One or more elements of the model can be selected by the user or programmatically by an app.

MarkusSteinbrecher, 2022-11-09:
@Adrian62D here too, how is this different to #475 ?

Adrian62D, 2022-11-09:
#475 is about allowing the user to actually filter through the elements.

In BimCollab this functionality is called SmartViews:
https://www.bimcollab.com/en/products/bimcollab-zoom/smart-views

OlegMoshkovich, 2023-01-11:
Does this issue include selecting multiple elements in the viewport and the tree?

Adrian62D, 2023-01-11:
This issue covers:
- viewport (mouse)
- programmatic access via matrix

The tree view is covered in #417.
Should we merge these two issues?

OlegMoshkovich, 2023-01-11:
[image]

Adrian62D, 2023-01-12:
@OlegMoshkovich Based on our last discussion we should provide smaller, isolated PR's. Accordingly I've updated the description of this issue to change the scope to viewport only - without programmatic access. The PR for the implementation follows shortly.

- Programmatic access (matrix widget api) has now been split to: #541.
- Has there already been an issue created for the selection-sets idea? This should be splitted as well.

OlegMoshkovich, 2023-01-12:
I assumed the point of selecting multiple elements is to control the visibility of what is selected.

pablo-mayrgundter, 2023-01-12:
We're having a design discussion in this thread:

https://discord.com/channels/853953158560743424/1063129406563950682

We're thinking to have some local lookup tables in a state file stored in the client and serialize around matrix.  This will reference IFC UUIDs and expressIDs as serialized at a particular version of the model.  If the client sees a changed model version or changed state file version, they rebuild from scratch.  Otherwise, we'll have some ability to use the IDs stored in the table as stable keys for shared state between clients.

OlegMoshkovich, 2023-02-09:
@Adrian62D is this issue done?

Adrian62D, 2023-02-09:
Remaining work covered in #548

closed 2023-02-09 by Adrian62D