bldrs-ai/Share #387: Color mapping support

Adrian62D, 2022-09-15:
Color mappings for values found in the attributes can be defined (“heatmap”, ok if restricted to admin in the beginning)

MarkusSteinbrecher, 2022-09-15:
how is this different to [383](https://github.com/bldrs-ai/Share/issues/383)?

rti-vyzn, 2022-09-16:
> how is this different to [383](https://github.com/bldrs-ai/Share/issues/383)?

I'm not entirely sure if I understand the issue correctly, but I know that in other viewers (e.g. xeokit) you can pre-assign colours to different IFC types, in order to get them right in the first render. 

Otherwise I'm not entirely sure if the entire scene would have to be in place in order to assign colours to specific elements - meaning it would not have the desired colours from the beginning. 

But maybe this is not what we're talking about, I don't know.

Adrian62D, 2022-09-16:
Sorry, this is a duplicate, you may close it.

closed 2022-09-16 by Adrian62D