bldrs-ai/Share #388: multiple models can be loaded

Adrian62D, 2022-09-15:
Multiple models can be loaded

- multiple models in the same scene (1 coordinate system)
- multiple models side-by-side (multiple coordinate systems)

MarkusSteinbrecher, 2022-09-15:
[duplicate](https://github.com/bldrs-ai/Share/issues/248), please close

closed 2022-09-15 by MarkusSteinbrecher

Adrian62D, 2022-09-15:
Does  #248 support side-by-side? I think it only covers a single scene scenario.

Imagine you want to visually compare the heatmap of version x and version y side-by-side.

MarkusSteinbrecher, 2022-09-15:
I think it does, otherwise let's add it to the description of 248.

---
We currently see the following use cases that require loading multiple IFC into a scene:

Engineering disciplines: as an architect or engineer I want to be able to load multiple models, e.g. the architecture model and the HVAC model into one scene to compare them visually
Big projects: as an architect I want to load multiple models of one project, i.e. models of multiple houses into the scene to see the whole project
Versioning: as an architect or engineer I want to load different versions of a model into a scene to compare them visually