bldrs-ai/Share #391: UI configuration

Adrian62D, 2022-09-15:
Regarding the viewer itself: 
allow to hide or show different UI elements, such as the tree view, the share buttons, the search bar…

Available to the user and also programmatically via app.

pablo-mayrgundter, 2022-09-15:
We have been configuring the components on a kind of "scene" basis:
  - open side-drawer and move operation icons when a drawer-based component is active
  - when on mobile, allow toggling of nav tree
  
@MarkusSteinbrecher has proposed a Google Apps-like reveal.  I see this breaking into:
  0. App not visible
  1. Small docked icon, minimal state (e.g. icon with red notifications badge)
  2. Panel, e.g. like current Issues and Properties tools
  3. Full-screen

The benefit of this approach is that the UX interaction between components and apps can be designed to support a few layouts.  I'd compare this to G Apps, Figma.

I'd contrast this with generally configurable UIs like an IDE such as Blender, IntelliJ.  We kinda want to avoid this as UI complexities is one of the key problems in existing AEC desktop tools.

Adrian62D, 2022-12-09:
@pablo-mayrgundter We'll start working on this. Could you please assign it to me? Thank you.

pablo-mayrgundter, 2022-12-22:
Moving discussion here from Ibrahim's PR... mostly looks good.  I'm not sure it makes sense to have each button configurable.  Are there any supporting use-cases for turning each one on or off?

Alternatively, how about a single toggle for the entire OperationGroup, or functional subsets that we define here first.

Adrian62D, 2022-12-22:
The use cases we currently see are as follows:

- cross mark -> to be hidden
- question mark -> sometimes visible / sometimes hidden

[image]

So this seems to fit well to the current structure in the UI. This means the controls could be grouped as follows:

- SearchBar
- TreeView
- Collaboration (button group 1)
- ModelInteraction (button group 2)
- Settings (button group 3)

Would this approach be OK for you?

PR #525 (2023-02-01): VYZN: Viewer: UI configuration via Matrix config message
This PR refactors the underlying classes for the widget API communication mechanism that allows the Share app to be treated as an iframe-embeddable app that supports receiving and sending messages back and forth with the embedding apps. 

The main theme for this PR is supporting #391 **UI configuration** via this previously mentioned widget API mechanism.
 
The PR also stabilizes the e2e tests for this API integration and adds the required test for the new functionality **UI configuration**.

09/01 Update:  Closes #391, closes #515, clos [... cut]
closed 2023-02-01 by pablo-mayrgundter