bldrs-ai/Share #392: API for apps to share state, events and control

Adrian62D, 2022-09-15:
Apart from that, the viewer would have to expose some of its state to its context - we somehow need to subscribe to changes in the public state of the viewer, in order to sync the 3D view and whatever other view is happening in the context. For example: user selects an item in the viewer, it should also be selected in a list, or it should allow us to display some contextual menu to be able to perform some of the actions described earlier - hiding that element, or including it in a group, or zooming into it. Or, a completely different case: the user wants to load a model, but it’s huge and it takes a long time - the context might want to know when and if the model was actually loaded, or if there was an error while loading it.

The utilities of[ xeokit-sdk](https://xeokit.io/) are a good reference because even though it’s a bit old-fashioned

pablo-mayrgundter, 2022-09-15:
I'm not sure that Viewer is the right terminology here.  I think you're probably referring to Share, this whole product?

In your example "user selects an item in the viewer, it should also be selected in a list, or it should allow us to display some contextual menu", I think this is just IFC.js for now.  We're considering a pluggable viewer but that's not in scope right now.

In the second example, Share does have this state and can share it in an API.  Please refer to:

https://github.com/bldrs-ai/Share/blob/main/src/Containers/CadView.jsx#L83

There's 4 hooks there for progressive stages of the model load.  I think these are probably the stateful variables you're after?

I'm considering using the https://github.com/matrix-org/matrix-widget-api/ for structuring the messaging between Share and Apps.  How does it look to you?  We'd pass these vars via this API's messages.  You'd implement the Widget side on Vyzn, we'd implement the Container side on Share.

pablo-mayrgundter, 2022-09-15:
Per discussion, let's track all App API requirements here.  Changing the title to remove "Viewer"

Adrian62D, 2022-09-16:
Hi Pablo, I copied this text over from the previous requirements doc. 

I initially thought there might be two API's being available to an app hosted on Bldrs, one api for basic integration (persistence?) and then one for the viewer. Yes I am refering to Share as a whole. If it is a single API that's fine as well. Also I thought that maybe the viewer could be an app with intents in the future (splitting viewer/platform in Share). But that is probably for later.

I'll take a look at the matrix widget api. Thank you!