bldrs-ai/Share #398: iframe integration issues

Adrian62D, 2022-09-19:
Issue analysis from Rafael:

_I saw the [prototype](https://deploy-preview-379--bldrs-share.netlify.app/share/v/p/index.ifc#c:-111.37,14.94,90.63,-43.48,15.73,-4.34) of bldrs including vyzn in an iframe. As you might have noticed it’s throwing an error: Refused to display 'https://www.vyzn.tech/' in a frame because it set 'X-Frame-Options' to 'sameorigin'.
From  your side, I think you should change the domain you’re trying to display from ‘https://www.vyzn.tech/’ to ‘https://dev.vyzn.tech/’
From our side, we need to reconfigure the server to add source (https://deploy-preview-379--bldrs-share.netlify.app/?) to the Content-Security-Policy header in the response._

MarkusSteinbrecher, 2022-09-19:
@pablo-mayrgundter can you change the domain name as stated above?

Adrian62D, 2022-09-19:
Basic loading of the page within iframe works already on my end when changing the URL:

[image]

Adrian62D, 2022-09-19:
@MarkusSteinbrecher Does this work for you as well?

MarkusSteinbrecher, 2022-09-19:
Yes, I see the same screen. Clicking on "Anmelden" does not do anything though

Adrian62D, 2022-09-19:
Yes, I assume the API calls do fail due to the missing Content-Security-Policy.

pablo-mayrgundter, 2022-09-20:
@MarkusSteinbrecher I changed the target of the PR, but to a locally hosted Matrix Widget, as I need to get that working first.  NOTE: clicking the Matrix buttons sets off an infinite loop of widget-container msgs.  Working on that now