bldrs-ai/Share #406: Text in Notes window is cut at both sides

MarkusSteinbrecher, 2022-09-26:
Some users have an issue with notes display:

[image]

Chrome version: Version 105.0.5195.125 (Offizieller Build) (x86_64)

pablo-mayrgundter, 2022-09-26:
Could you post a link?  Can't reproduce.  Desktop?

MarkusSteinbrecher, 2022-09-26:
https://bldrs.ai/share/v/gh/Swiss-Property-AG/Turmfalke-Public/main/TURMFALKE.ifc#i:1340200278::c:35.31,67.23,58.84,-2.77,7.81,15.5

Desktop

pablo-mayrgundter, 2022-09-26:
Markus can repro in Chrome/Safari and FF.

I can't repro in those.

We don't have an IE tester.

OlegMoshkovich, 2022-09-26:
Look ok for me.

MarkusSteinbrecher, 2022-09-26:
fixed with #407

OlegMoshkovich, 2022-09-26:
[image]

Steps to reproduce - set the chrome default font to small in settings

MarkusSteinbrecher, 2022-09-27:
Looking good

closed 2022-09-27 by MarkusSteinbrecher