bldrs-ai/Share #429: Cut plane makes model disappear for Schep

pablo-mayrgundter, 2022-10-07:
https://bldrs.ai/share/v/gh/IFCjs/test-ifc-files/main/Schependomlaan/IFC%20Schependomlaan.ifc#c:60.45,-4.32,60.59,1.17,5.93,-3.77

Works for L0 and L1, but higher stories makes model disappear.  I'm guessing this is just incorrect floor finding, as L1 is the whole top of the building.

@mkeshavarzi Can you take this?

There's also still a console log being spit out

[image]

mkeshavarzi, 2022-10-09:
@pablo-mayrgundter  - This seems to be an error with the file, where the author has defined levels in which are not used. We may want to filter it, but during one of our all-hands meetings, we decided not to intervene with author mistakes in files.

The console issue is being addressed in #424.

mkeshavarzi, 2022-11-03:
Addressed in #456.

pablo-mayrgundter, 2023-01-26:
To clarify what's going on, it's because IFC files have a length unit variable that we ignore.  we assume all units are in meters, but that's just common in IFC.  Schep uses millimeters, so our attempt to split the floors based on their height is way off.

The fix should read the IFC unit element and also include e2e testdata tests that validate with IFC files of different length units.

OlegMoshkovich, 2023-02-03:
makes sense - Should the title of the issue reflect your comment?

pablo-mayrgundter, 2024-03-08:
We removed this component, and cutplanes work for this case.

closed 2024-03-08 by pablo-mayrgundter