bldrs-ai/Share #441: System: FE: Interactions with a model are slow, if they require tree traversal

MarkusSteinbrecher, 2022-10-18:
When a user interacts with a model, e.g. searches or selects elements, the UI interaction is sloe (depending on the tree a couple to multiple seconds). This is independent on the actual model size but rather the tree traversal that takes too long.

OlegMoshkovich, 2024-07-16:
@pablo-mayrgundter this a unacceptable UX, and creates a bad impression of the product.
From my POV it shall be fixed asap, specifically if we are planning to expose 'Share' to Nokera.
Thoughts?

OlegMoshkovich, 2024-07-17:
New Issue is created to implement the initial fix of disabling the auto activation of the tree.
When the tree is activated for the first time, create a snackbar message to the user:
'Tree selection is in progress feature. We know it is slow. The fix is coming soon.