bldrs-ai/Share #442: Viewer:  Add Navigation Cube

OlegMoshkovich, 2022-10-18:
Navigation cube is a viewer camera controller that allows the user to quickly switch between basic predefined views.

- Create a three.js navigation cube.
- Place navigation cube in the bottom right corner.
- Add a switch between orthogonal and perspective camera views.

OlegMoshkovich, 2022-10-18:
[image]
It seem the camera is not aligned between the cube and the model.

OlegMoshkovich, 2022-10-18:
[image]

when back side of the cube is selected the model zooms in way to close.

RyugaRyuzaki, 2022-10-18:
1-It seem the camera is not aligned between the cube and the model.
2 -when back side of the cube is selected the model zooms in way to close.
is there a way to control the camera zoom level
=> I found this issue, i did not transform to center of model. 
exam :  back side
           coords.x = -r;=>must be -x+c.x
	   coords.y = c.y;
	   coords.z = c.z;
=> i will fix them now

OlegMoshkovich, 2022-10-18:
[image]
[image]

Hmm hovering over the face of the cube doe not always selects the surface.

RyugaRyuzaki, 2022-10-18:
yeah, well, may be i will push z-index of container of cube to highger than main view

OlegMoshkovich, 2022-10-18:
ah maybe that is it - but that would mean that something in the way and blocking it, not sure that is the problem -- but maybe

OlegMoshkovich, 2022-10-18:
[image]

actually you are right - only selection in the green box is allowed -- anything outside of the green box is not registering

RyugaRyuzaki, 2022-10-18:
@OlegMoshkovich you were right, function raycaster is wrong. i will fix it.

OlegMoshkovich, 2022-10-18:
awesome!

RyugaRyuzaki, 2022-10-18:
[image]
 => i found the issue. Because we take zIndex of NavPanel higher than viewer-container
=> so i just change the cube style.zIndex=999=> it works
=> keep fixing the camera and create a toggle button camera change type

RyugaRyuzaki, 2022-10-18:
=>fixed cast element when client click on buce to change camera
const filterElementClick = _this.scene.children.filter((child) => {
			return child.userData.Element;
		});
and : const intersects = rayCaster.intersectObjects(filterElementClick);
:(( => i try to create a button change camera type, it seems to does not work

OlegMoshkovich, 2022-10-20:
super cool!

OlegMoshkovich, 2023-01-19:
Closed PR is here 
https://github.com/bldrs-ai/Share/pull/435

closed 2024-08-12 by OlegMoshkovich