bldrs-ai/Share #453: Method to lookup EID (expressID) by GID

clecherbauer, 2022-10-25:
Hi, to finish up the integration of bldrs into vyzn i need a function in bldrs that can lookup expressID's by GID because vyzn only stores GID's.

pablo-mayrgundter, 2022-11-02:
Hey Christian, I didn't realize this, but I put in single GUID search in the very beginning:

https://bldrs.ai/share/v/p/index.ifc?q=02uD5Qe8H3mek2PYnMWHk1

Just use ?q=<GUID> on the URL.

Updated our docs:
https://github.com/bldrs-ai/Share/wiki/Design:-URL-Structure#guid-search

Is that sufficient, or are you wanting to get the EID back that Bldrs' model is using?

clecherbauer, 2022-11-07:
@pablo-mayrgundter highlighting of one element works that way. We still need bldrs to trigger an event / update the store if some element is highlighted inside bldrs.

pablo-mayrgundter, 2022-11-07:
Ok, that also is done already, see:

https://github.com/bldrs-ai/Share/blob/main/src/Containers/CadView.jsx#L72

and below that a few lines is selectedElements as well.  This is how we currently do multi-select