bldrs-ai/Share #475: As a user I can filter one or more elements

Adrian62D, 2022-11-07:
As a user I can filter one or more elements

Part of [BimTool](https://docs.google.com/presentation/d/1hwNEDTwzOExQwTuTr3-DNgs4JD8ijuOHu77AHMpEI20/edit#slide=id.g14aa60b89b7_0_92)

MarkusSteinbrecher, 2022-11-09:
Bldrs supports selecting multiple elements already but only through search. This features enables a user (or an app through api syntax) to select multiple elements by either: (1) shift+click (lmb) multiple elements in the viewer or (2) shift+click multiple elements in tree view.

MarkusSteinbrecher, 2022-11-10:
as a first pass we could implement filtering through search, similar to how Github does it "is:open is:issue author:MarkusSteinbrecher"

Adrian62D, 2023-01-10:
@pablo-mayrgundter What are your thoughts on this?

Adrian62D, 2023-02-03:
Consider IDS for selection.