bldrs-ai/Share #490: NavTree isn't scrolling

pablo-mayrgundter, 2022-11-17:
When tree scrolls off the page, I can't gesture scroll down in OSX.  @MarkusSteinbrecher noted this and I think has a mouse wheel.  So probably not an input device thing.

[image]

closed 2022-11-17 by OlegMoshkovich

pablo-mayrgundter, 2022-11-17:
Related, comment scroll is broken.. will just reuse this bug.

MarkusSteinbrecher, 2022-11-17:
Scrolling the tree works again. Mobile seems to be broken now. The tree is cut in the middle. And moving the model does only work on the lower part of the screen. If you try it in the middle, the model is not moving (Chrome and Safari). Try loading the homepage on mobile and move the model in the middle.

[image]

PR #491 (2022-11-17): NavPanel scroll  fix
This PR fixes #490.