bldrs-ai/Share #494: Bug: Property window moves down when "show all properties" is selected

MarkusSteinbrecher, 2022-11-24:
When opening the properties panel and clicking on the "show all properties" radio button the content moves down like this:

[image]

This seems to happen with every model but is worse if the properties content is longer (i.e. moving down further).

pablo-mayrgundter, 2022-11-25:
Looks like broke in #486:

https://github.com/bldrs-ai/Share/blob/02e2fc31c9516818aa92ec0b525aab2617bd1666/src/Components/SideDrawer.jsx#L65

I replaced the makeStyle css with sx, but that's only supported on Box.

pablo-mayrgundter, 2022-11-29:
@MarkusSteinbrecher all good?

MarkusSteinbrecher, 2022-11-29:
Looks good

closed 2022-11-29 by MarkusSteinbrecher