bldrs-ai/Share #497: Cutting plane orientation does not work reliably

Adrian62D, 2022-12-02:
Steps to reproduce
1. Navigate to [Eisvogel](https://bldrs.ai/share/v/gh/Swiss-Property-AG/Eisvogel-Public/main/EISVOGEL.ifc#c:107.36,8.46,156.67,3.52,2.03,16.71)
2. With your cursor hover over the roof:
[image]
3. Press Q

**Expected**
The cutting plane is created taking into account the surface normal of the roof
[image]

**Actual**
The cuttin plane does not take into account the surface normal of the roof
[image]

Relates to #332

Adrian62D, 2022-12-02:
Example from BimCollab:

[image]

MarkusSteinbrecher, 2022-12-02:
There is only xyz cutting planes at the moment, so it works as intended. This would be a feature request I guess.

Adrian62D, 2022-12-02:
It seems, orientations based on surface normals are already supported for *walls*. See here, it works:

[image]

I renamed the title as feature request.

closed 2022-12-02 by pablo-mayrgundter