bldrs-ai/Share #518: Rendering of individual nodes

Adrian62D, 2022-12-21:
Technical changes:
- Currently the rendering happens per-model 
- This needs to be changed so that individual nodes can be rendered

Option 1:
- Implement this change as a PR towards IFCJS

Option 2:
- Implement this change by controlling the rendering in Bldrs itself (three.js)

This is a precondition to #383, #384, #475.

Adrian62D, 2022-12-21:
@pablo-mayrgundter Is there a preference for Option 1 or 2?

Adrian62D, 2022-12-22:
We'll provide a more detailed assessment on this issue soon.

Ibrahim5aad, 2022-12-26:
There are two paths that could be followed. Implementing this as a PR towards IFC.js is tricky for multiple reasons but the most  important reason is that the current IFC.js per-model rendering in my opinion is a design choice to reduce the number of draw/render calls to the GPU; one draw call per model. And a lot of their Three.js abstractions and wrappers depend on this design.

To overcome the drawbacks of having one mesh per model, they offer what is called subsets operations which allows for creation of subsets of the model mesh by creating new geometry that copies a specific range of indices from the combined model indexed mesh. 

This subsets operations feature is one way we could follow to achieve what we want. [Option 1 in the referenced image].
We let IFC.js load/parse the geometries but interrupt their pipeline at the point they add the mode mesh to the scene, and create a subset per IFC element and add those created subsets to the scene instead.

Second option is about overcoming one drawback for the first option; those subsets are already identical meshes to the elements meshes created while IFC.js stream the loaded buffer geometries, but we don't take advantage of those created geometries, they will be disposed after combining them into the model mesh. So instead of interrupting the pipeline after those meshes are merged, we collect them before they are merged.  

Both option will require to break some IFC.js abstractions and some of their three.js wrappers. It's unfortunate that the library don't offer much customizations to how their parsing-meshing-rendering pipeline is assembled. But it shouldn't be a complicated task, the examples already gives glimpses of how to do part of we are targeting, like using subsets to hide elements (of course with breaking some of their out-of-the-box features like selection). 

[image]

Adrian62D, 2023-01-03:
@Ibrahim5aad Thank you!
@pablo-mayrgundter Do you have a preference regarding these approaches?

pablo-mayrgundter, 2023-01-05:
Hi @Ibrahim5aad, thanks for the writeup!

I am a bit confused.  I have seen this subset API:

https://ifcjs.github.io/info/docs/Guide/web-ifc-three/Tutorials/Highlighting/

Are you aware of this?  Could you elaborate on why this is insufficient and why you think a PR to ifcjs is needed?

Ibrahim5aad, 2023-01-05:
Hi @pablo-mayrgundter, actually a PR to ifcjs was not my idea, and I don't think it's required at all.
Subset operation is indeed can be sufficient (with side-effects) to implement the behaviour wanted (it's the first option I'm mentioning), but with breaking some of the higher level abstractions (the ifcjs examples clearly shows naïve implementation of this.)

Adrian62D, 2023-01-06:
I updated the issue description. No PR to ifcjs needed.

OlegMoshkovich, 2023-01-11:
So is the current plan to proceed with Subset operations as described here https://ifcjs.github.io/info/docs/Guide/web-ifc-three/Tutorials/Highlighting/?

Adrian62D, 2023-01-11:
This is to be decided. Both options seem similar. Do you have a preference?

OlegMoshkovich, 2023-01-11:
Whatever is the simplest way forward. 
When the data model work is complete we might have to rewrite the feature.

closed 2023-05-03 by Adrian62D