bldrs-ai/Share #602: Error on opening a privately hosted model

MarkusSteinbrecher, 2023-02-14:
Trying to open an ifc file that is hosted on a private Github repository.

First I got an error message about OAuth access from third party apps and I changed the settings of the organization on Github:   

[image]

Login using Github credentials works fine but I get the following error trying to load the ifc:

[image]

Does opening privately hosted ifc models work for you?

Thanks!

pablo-mayrgundter, 2023-02-15:
Huh.. yeah, I can open e.g. https://github.com/pablo-mayrgundter/private/blob/main/haus.ifc by dropping it into bldrs when i'm logged in.

I wonder if this is a cross-org thing, as I'm the owner of my own repo

@oo-bldrs thoughts?

MarkusSteinbrecher, 2023-02-15:
I own that repo too. I added both of you as admins so you can check if you can open: https://github.com/Swiss-Property-AG/Inspire-Better-Living/blob/main/2202_ARC-BadWiessee_221208%20(1).ifc

Thanks

pablo-mayrgundter, 2023-02-15:
Ok yeah, that fails for me too

oo-bldrs, 2023-02-15:
I don't have access with my username, @oo-bldrs.

oo-bldrs, 2023-02-15:
@pablo-mayrgundter One thing to keep in mind regarding the GitHub workflow for collaborators (which is expressly different than team members): @MarkusSteinbrecher added us as collaborators. This requires you to explicitly accept the invite first *before* you can access the repository's contents.

The issue here is double quoting. If you paste the raw URL into the search bar, the `%20(1).ifc` portion is encoded by Javascript into `%2520%281%29.ifc`. This is a completely different (and wrong) filename which results in a 404 Not Found from GitHub.

MarkusSteinbrecher, 2023-02-15:
Cool, it works: https://github.com/Swiss-Property-AG/Inspire-Better-Living/blob/main/BadWiessee.ifc

Could have sworn I used the same file on public but obviously I didn't.

Thanks! Private hosting, wohoo! ;)

pablo-mayrgundter, 2024-03-08:
Fixed

closed 2024-03-08 by pablo-mayrgundter