bldrs-ai/Share #619: Element selection performance sub par

Adrian62D, 2023-02-20:
Steps to reproduce:

1. Select Seestrasse Model from the load dialog
2. Change element selection by clicking on different walls

[image]

Expected:
Selection changes in <50ms

Actual:
Selection changes in approx 2 seconds

It seems that element selection performs way better on smaller models ("Momentum").

Ibrahim5aad, 2023-06-11:
@Adrian62D  I found out why this happens. It's not the selection itself it seems. Updating the TreeView expanded elements is blocking the app for some significant milliseconds. This update is part of the selected elements effect. If you tried to comment out those calls to set the expanded tree items, you won't encounter such lag even in bigger models.

[image]

Adrian62D, 2023-06-12:
@Ibrahim5aad Great, thank you! What options for action do we have? Could the handling of the three view be done without blocking the ui?