bldrs-ai/Share #637: Loading private model works when copying GH-link but not when forwarding the link

MarkusSteinbrecher, 2023-02-28:
We have a privately hosted model that can be viewed in bldrs when copying the gh-link into the search bar:

[image]

but when I share the link I get the following error: 

[image]

[image]

I am happy to give you access to the Repo but I was able to reproduce the same issue with another model in another private Repo.

Thanks!

pablo-mayrgundter, 2023-03-07:
Hey O, I lost track and was going to start work on this.. assigning you since you're on it.

MarkusSteinbrecher, 2023-03-14:
any update on this?

OlegMoshkovich, 2023-03-20:
@oogali is #630 addressing this issue?

pablo-mayrgundter, 2024-03-08:
Fixed

closed 2024-03-08 by pablo-mayrgundter