bldrs-ai/Share #686: Public model opens fine when logged in but not without being logged in

MarkusSteinbrecher, 2023-03-27:
Trying to open this [model](https://bldrs.ai/share/v/gh/Swiss-Property-AG/Unteraegeri-Public/main/Unteraegeri.ifc) which is hosted by Github LFS and it works fine if I am logged in on Bldrs but throws an error message if I try to open it without being logged in:$

[image]

[image]

Any idea? Am I doing something wrong or is this a bug? 

Thanks!

OlegMoshkovich, 2023-03-27:
Same for me, getting the error when not logged in, but can get to it through the file browser.
[image]

oo-bldrs, 2023-03-27:
@OlegMoshkovich @pablo-mayrgundter In order to address this, we need to come to a decision on how to handle specific, read-only GitHub API operations for application users who are not logged in, given that an individual's limit is no more than 60 requests per hour.

Also, this issue is a duplicate of the original GitHub LFS issue (#277).

pablo-mayrgundter, 2023-03-27:
This file is only 35MB, so LFS not needed.  Workaround for now is to recreate the repo without LFS.

[image]

OlegMoshkovich, 2023-03-27:
Created a test repo with the file.
It opens ok for me.
https://bldrs.ai/share/v/gh/OlegMoshkovich/Unteraegeri--test/main/Unteraegeri.ifc#c:165.955,122.104,149.039,37.207,-6.643,20.291

@MarkusSteinbrecher please test.

MarkusSteinbrecher, 2023-03-27:
Thanks! It works fine without LFS and for public. The reasoning behind LFS is, that we will need LFS for every project at some point so it is easier to setup the Github Repo for LFS when it is created. We can work around that for most models for now but this one here is very hard to make small for example: https://bldrs.ai/share/v/gh/Swiss-Property-AG/Eggstock/main/Eggstock_INTERN.ifc (still private but I can give you access). So, would still be great if that issue is solved.

oo-bldrs, 2023-05-03:
@MarkusSteinbrecher Do you have a URL to a model that's publicly available in a Git LFS-enabled repository?

MarkusSteinbrecher, 2023-05-04:
Not at the moment ... do you want me to create one?

oo-bldrs, 2023-05-04:
@MarkusSteinbrecher I would appreciate it, if you could.

I have a potential fix and I'd like to test before we claim success.

MarkusSteinbrecher, 2023-05-04:
here you go: https://github.com/Swiss-Property-AG/Turmfalke-Test/blob/main/Turmfalke-Test.ifc

and it opens fine if logged in and doesn't if not.

oo-bldrs, 2023-05-04:
The potential fix seems to work...

https://deploy-preview-717--bldrs-share.netlify.app/share/v/gh/Swiss-Property-AG/Turmfalke-Test/main/Turmfalke-Test.ifc

MarkusSteinbrecher, 2023-05-04:
Yes! You can claim success :)

closed 2023-05-04 by oo-bldrs

oo-bldrs, 2023-05-04:
Pushed live: https://bldrs.ai/share/v/gh/Swiss-Property-AG/Turmfalke-Test/main/Turmfalke-Test.ifc