bldrs-ai/Share #1041: Camera position restore triggered by Share control button, Safari only

pablo-mayrgundter, 2024-03-12:
In Safari, if the page is loaded with a camera hash state, going to share a different camera position will restore the original state.  This is in Safari only.

pablo-mayrgundter, 2024-03-27:
@OlegMoshkovich also bouncing back.. can't repro here

bldrs.ai camera sharing
OSX 14.3.1
  - [x] Brave 1.64.109
  - [x] Safari 17.3.1
  - [x] FF 124

OlegMoshkovich, 2024-03-28:
Setup: Chrome Version 122.0.6261.94 (Official Build) (x86_64)

Model Link:
https://bldrs.ai/share/v/gh/Swiss-Property-AG/Momentum-Public/main/Momentum.ifc#c:-38.64,12.52,35.4,-5.29,0.94,0.86;

Steps to Reproduce:
- [ ] **Action:** Load the Bldrs
- [ ] **Action:** Load Momentum
- [ ] **Action:** Move the Camera
- [ ] **Action:** Click on Share Control
- [ ] **Response:** Camera coordinated are captured by the URL
- [ ] **Action:** Close the share dialog
- [ ] **Action:** Move the model
- [ ] **Action:** Click on the cutplane control
- [ ] **Action:** Select the plan cutplane
- [ ]  **Expected Response:** The camera position remains the same and cutplane is place along the appropriate axis
- [ ]  **Actual Response:**  The camera bounces back to the coordinates captured by the URL

OlegMoshkovich, 2024-03-28:
@pablo-mayrgundter PTAL

pablo-mayrgundter, 2024-04-02:
Actually happening in Brave too.. i think it's not a share problem but rather cutplane fighting with share camera params

pablo-mayrgundter, 2024-04-11:
Fixed in https://github.com/bldrs-ai/Share/pull/1108

closed 2024-04-11 by pablo-mayrgundter