bldrs-ai/Share #1068: Title: AI Render Only Captures View Correctly on First Attempt; Subsequent Captures Fail

OlegMoshkovich, 2024-03-23:
When using the AI render feature, the initial capture of the view is successful and behaves as expected. However, any attempts to capture the view after the first one result in a failure. The AI render does not seem to update or capture the new view correctly on subsequent attempts.

pablo-mayrgundter, 2024-03-27:
Didn't repro on first checkout.  Here's Brave prod.. will check safari next

1st screen, after render
[image]

2nd time, new prompt
[image]

pablo-mayrgundter, 2024-03-27:
Same with Safari 17.3.1 and FF 124 on OSX 14.3.1

Oleg, can you repro and document your setup?  Will reassign to you now

OlegMoshkovich, 2024-03-28:
Steps to Reproduce:
- [ ] **Action:**Load the Bldrs
- [ ] **Action:**Load Momentum
- [ ] **Action:**Click Magic wand
- [ ] **Response:** The correct view is captured
- [ ] **Action:**Close imagine dialog
- [ ] **Action:**Move the camera
- [ ] **Action:**Click Magic wand
- [ ]  **Expect:** The correct view is captured
- [ ]  **Actual:** Previous view is captured by the imagine dialog

https://bldrs.ai/share/v/gh/Swiss-Property-AG/Momentum-Public/main/Momentum.ifc#c:-38.64,12.52,35.4,-5.29,0.94,0.86;

pablo-mayrgundter, 2024-04-12:
Fixed in #1110

closed 2024-04-12 by pablo-mayrgundter