bldrs-ai/Share #1237: Refactor Load Model / OPFS / file logic

nickcastel50, 2024-07-16:


nickcastel50, 2024-07-17:
@pablo-mayrgundter

pablo-mayrgundter, 2025-03-13:
From discussion today, here's an update to the download flow that will save small content responses directly, to avoid a second download request, and also clarifies our handling of local cache state in OPFS:

1. first time
  1. /contents
    1. For small files, maybe just use the content
    2. download OTURL, story <sha>.ext
      1. Browser context keeps cache API to store etag data
2. Next time
  1. REQ /contents w/etag
    1. 304 Not Modified, no body, but we have a cached RSP
       1. Use cached SHA to see if file exists in OPFS (now we're safe with clearOPFSCache); if so, use it no DL
       2. If not GET /contents w/o ETAG
         1. For small files, maybe just use the content

pablo-mayrgundter, 2026-05-05:
Superseded by the new wiki page: **[Design: Model Hosting](https://github.com/bldrs-ai/Share/wiki/Design:-Model-Hosting)**.

That page covers the current direct-from-browser architecture (the rawgit.bldrs.dev proxy was retired in #1495), the motivating 2026-05-01 incident, why the LFS Range probe the proxy did is no longer needed (the GitHub Contents API returns an LFS-aware `download_url` directly), code flow, caching, and test infrastructure.